xboot/xfel #2: ERROR: Can't found any FEL device

Yangxiu123321, 2021-05-26:
执行时找不到设备
> mod@archlinux xfel git:[master] $ ./xfel version
ERROR: Can't found any FEL device

下面的结果返回NULL
`ctx.hdl = libusb_open_device_with_vid_pid(NULL, 0x1f3a, 0xefe8);`

查看usb设备是存在的
> mod@archlinux xfel git:[master] $ lsusb
Bus 001 Device 002: ID 8087:8001 Intel Corp. Integrated Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 003: ID 04f2:b502 Chicony Electronics Co., Ltd Integrated Camera
Bus 002 Device 002: ID 8087:07dc Intel Corp. Bluetooth wireless interface
Bus 002 Device 008: ID 1f3a:efe8 Allwinner Technology sunxi SoC OTG connector in FEL/flashing mode
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

平台：
archlinux
libusb 1.0.24-2: https://archlinux.org/packages/core/x86_64/libusb/

jianjunjiang, 2021-05-26:
用sudo权限执行，普通用户没有权限访问这个usb 设备。

Yangxiu123321, 2021-05-26:
问题已解决

closed 2021-05-26 by Yangxiu123321