xboot/xfel #6: support for MX35LF2GE4AD

orangecms, 2021-07-23:
Currently, this is what I get on my Nezha board:

```sh
$ xfel version
AWUSBFEX soc=0x00185900(D1) 0x00000001 ver=0x0001 0x44 0x08 scratchpad=0x00045000
$ xfel sid
d0001200c8145c14558181811ac82ccb
$ xfel spinand
Not found any spi nand flash
$ xfel spinor
The spi nor flash '0xffc226' is not yet supported
Not found any spi nor flash
```

I think it is actually a NAND though, as per
https://www.mxic.com.tw/en-us/products/NAND-Flash/Serial-NAND-Flash/Pages/spec.aspx?p=MX35LF2GE4AD&m=Serial+NAND&n=PM2794 (I read the name from the chip itself).

jianjunjiang, 2021-07-28:
Support spi nand must writing some code to spinand.c， don't modify code of spinor.c

jianjunjiang, 2021-12-11:
XFEL tools are very supportive of the needs of spi nor flash, but the support of spi nand flash has not been on the agenda. Considering that there are more and more players in F133 and D1, the support of this spi nand flash is imminent. Tian’s development and design, the support of spinand flash is basically completed.

Below is the list of supported spinand chips. I have tried my best to find the most complete list of chips. Of course, only the MX35LF2GE4AD chip has been tested, and the verification of other chips will be debugged after encountering specific problems.

```
	/* Winbond */
	{ "W25N512GV",       SPINAND_ID(0xef, 0xaa, 0x20), 2048,  64,  64,  512, 1, 1 },
	{ "W25N01GV",        SPINAND_ID(0xef, 0xaa, 0x21), 2048,  64,  64, 1024, 1, 1 },
	{ "W25M02GV",        SPINAND_ID(0xef, 0xab, 0x21), 2048,  64,  64, 1024, 1, 2 },
	{ "W25N02KV",        SPINAND_ID(0xef, 0xaa, 0x22), 2048, 128,  64, 2048, 1, 1 },

	/* Gigadevice */
	{ "GD5F1GQ4UAWxx",   SPINAND_ID(0xc8, 0x10),       2048,  64,  64, 1024, 1, 1 },
	{ "GD5F1GQ4UExIG",   SPINAND_ID(0xc8, 0xd1),       2048, 128,  64, 1024, 1, 1 },
	{ "GD5F1GQ4UExxH",   SPINAND_ID(0xc8, 0xd9),       2048,  64,  64, 1024, 1, 1 },
	{ "GD5F1GQ4xAYIG",   SPINAND_ID(0xc8, 0xf1),       2048,  64,  64, 1024, 1, 1 },
	{ "GD5F2GQ4UExIG",   SPINAND_ID(0xc8, 0xd2),       2048, 128,  64, 2048, 1, 1 },
	{ "GD5F2GQ5UExxH",   SPINAND_ID(0xc8, 0x32),       2048,  64,  64, 2048, 1, 1 },
	{ "GD5F2GQ4xAYIG",   SPINAND_ID(0xc8, 0xf2),       2048,  64,  64, 2048, 1, 1 },
	{ "GD5F4GQ4UBxIG",   SPINAND_ID(0xc8, 0xd4),       4096, 256,  64, 2048, 1, 1 },
	{ "GD5F4GQ4xAYIG",   SPINAND_ID(0xc8, 0xf4),       2048,  64,  64, 4096, 1, 1 },
	{ "GD5F2GQ5UExxG",   SPINAND_ID(0xc8, 0x52),       2048, 128,  64, 2048, 1, 1 },
	{ "GD5F4GQ4UCxIG",   SPINAND_ID(0xc8, 0xb4),       4096, 256,  64, 2048, 1, 1 },

	/* Macronix */
	{ "MX35LF1GE4AB",    SPINAND_ID(0xc2, 0x12),       2048,  64,  64, 1024, 1, 1 },
	{ "MX35LF1G24AD",    SPINAND_ID(0xc2, 0x14),       2048, 128,  64, 1024, 1, 1 },
	{ "MX31LF1GE4BC",    SPINAND_ID(0xc2, 0x1e),       2048,  64,  64, 1024, 1, 1 },
	{ "MX35LF2GE4AB",    SPINAND_ID(0xc2, 0x22),       2048,  64,  64, 2048, 1, 1 },
	{ "MX35LF2G24AD",    SPINAND_ID(0xc2, 0x24),       2048, 128,  64, 2048, 1, 1 },
	{ "MX35LF2GE4AD",    SPINAND_ID(0xc2, 0x26),       2048, 128,  64, 2048, 1, 1 },
	{ "MX35LF2G14AC",    SPINAND_ID(0xc2, 0x20),       2048,  64,  64, 2048, 1, 1 },
	{ "MX35LF4G24AD",    SPINAND_ID(0xc2, 0x35),       4096, 256,  64, 2048, 1, 1 },
	{ "MX35LF4GE4AD",    SPINAND_ID(0xc2, 0x37),       4096, 256,  64, 2048, 1, 1 },

	/* Micron */
	{ "MT29F1G01AAADD",  SPINAND_ID(0x2c, 0x12),       2048,  64,  64, 1024, 1, 1 },
	{ "MT29F1G01ABAFD",  SPINAND_ID(0x2c, 0x14),       2048, 128,  64, 1024, 1, 1 },
	{ "MT29F2G01AAAED",  SPINAND_ID(0x2c, 0x9f),       2048,  64,  64, 2048, 2, 1 },
	{ "MT29F2G01ABAGD",  SPINAND_ID(0x2c, 0x24),       2048, 128,  64, 2048, 2, 1 },
	{ "MT29F4G01AAADD",  SPINAND_ID(0x2c, 0x32),       2048,  64,  64, 4096, 2, 1 },
	{ "MT29F4G01ABAFD",  SPINAND_ID(0x2c, 0x34),       4096, 256,  64, 2048, 1, 1 },
	{ "MT29F4G01ADAGD",  SPINAND_ID(0x2c, 0x36),       2048, 128,  64, 2048, 2, 2 },
	{ "MT29F8G01ADAFD",  SPINAND_ID(0x2c, 0x46),       4096, 256,  64, 2048, 1, 2 },

	/* Toshiba */
	{ "TC58CVG0S3HRAIG", SPINAND_ID(0x98, 0xc2),       2048, 128,  64, 1024, 1, 1 },
	{ "TC58CVG1S3HRAIG", SPINAND_ID(0x98, 0xcb),       2048, 128,  64, 2048, 1, 1 },
	{ "TC58CVG2S0HRAIG", SPINAND_ID(0x98, 0xcd),       4096, 256,  64, 2048, 1, 1 },
	{ "TC58CVG0S3HRAIJ", SPINAND_ID(0x98, 0xe2),       2048, 128,  64, 1024, 1, 1 },
	{ "TC58CVG1S3HRAIJ", SPINAND_ID(0x98, 0xeb),       2048, 128,  64, 2048, 1, 1 },
	{ "TC58CVG2S0HRAIJ", SPINAND_ID(0x98, 0xed),       4096, 256,  64, 2048, 1, 1 },
	{ "TH58CVG3S0HRAIJ", SPINAND_ID(0x98, 0xe4),       4096, 256,  64, 4096, 1, 1 },

	/* Esmt */
	{ "F50L512M41A",     SPINAND_ID(0xc8, 0x20),       2048,  64,  64,  512, 1, 1 },
	{ "F50L1G41A",       SPINAND_ID(0xc8, 0x21),       2048,  64,  64, 1024, 1, 1 },
	{ "F50L1G41LB",      SPINAND_ID(0xc8, 0x01),       2048,  64,  64, 1024, 1, 1 },
	{ "F50L2G41LB",      SPINAND_ID(0xc8, 0x0a),       2048,  64,  64, 1024, 1, 2 },

	/* Fison */
	{ "CS11G0T0A0AA",    SPINAND_ID(0x6b, 0x00),       2048, 128,  64, 1024, 1, 1 },
	{ "CS11G0G0A0AA",    SPINAND_ID(0x6b, 0x10),       2048, 128,  64, 1024, 1, 1 },
	{ "CS11G0S0A0AA",    SPINAND_ID(0x6b, 0x20),       2048,  64,  64, 1024, 1, 1 },
	{ "CS11G1T0A0AA",    SPINAND_ID(0x6b, 0x01),       2048, 128,  64, 2048, 1, 1 },
	{ "CS11G1S0A0AA",    SPINAND_ID(0x6b, 0x21),       2048,  64,  64, 2048, 1, 1 },
	{ "CS11G2T0A0AA",    SPINAND_ID(0x6b, 0x02),       2048, 128,  64, 4096, 1, 1 },
	{ "CS11G2S0A0AA",    SPINAND_ID(0x6b, 0x22),       2048,  64,  64, 4096, 1, 1 },

	/* Etron */
[... 23 lines cut ...]
	{ "EM73E044SNA",     SPINAND_ID(0xd5, 0x03),       4096, 256,  64, 2048, 1, 1 },
	{ "EM73E044SND",     SPINAND_ID(0xd5, 0x0b),       4096, 240,  64, 2048, 1, 1 },
	{ "EM73E044SNB",     SPINAND_ID(0xd5, 0x23),       4096, 256,  64, 2048, 1, 1 },
	{ "EM73E044VCA",     SPINAND_ID(0xd5, 0x2c),       4096, 256,  64, 2048, 1, 1 },
	{ "EM73E044VCB",     SPINAND_ID(0xd5, 0x2f),       2048, 128,  64, 4096, 1, 1 },
	{ "EM73F044SNA",     SPINAND_ID(0xd5, 0x24),       4096, 256,  64, 4096, 1, 1 },
	{ "EM73F044VCA",     SPINAND_ID(0xd5, 0x2d),       4096, 256,  64, 4096, 1, 1 },
	{ "EM73E044SNE",     SPINAND_ID(0xd5, 0x0e),       4096, 256,  64, 4096, 1, 1 },
[... cut ...]
	{ "FM35Q1GA",        SPINAND_ID(0xe5, 0x71),       2048,  64,  64, 1024, 1, 1 },

	/* Paragon */
	{ "PN26G01A",        SPINAND_ID(0xa1, 0xe1),       2048, 128,  64, 1024, 1, 1 },
	{ "PN26G02A",        SPINAND_ID(0xa1, 0xe2),       2048, 128,  64, 2048, 1, 1 },

	/* Ato */
	{ "ATO25D1GA",       SPINAND_ID(0x9b, 0x12),       2048,  64,  64, 1024, 1, 1 },

	/* Heyang */
	{ "HYF1GQ4U",        SPINAND_ID(0xc9, 0x51),       2048, 128,  64, 1024, 1, 1 },
	{ "HYF2GQ4U",        SPINAND_ID(0xc9, 0x52),       2048, 128,  64, 2048, 1, 1 },
```