xboot/xfel #10: [编译错误] 发行版本内缺少相关文件

taotieren, 2021-11-09:
在最新的 v1.2.0 和 v1.2.1 发行版中，使用源码编译发现发行版包内缺少文件。详细如下
v1.2.0
```bash
➜  xfel-1.2.0 git:(master) ✗ tree
.
├── 99-xfel.rules
├── LICENSE
├── Makefile
├── README.md
├── chips
│   ├── a10.c
│   ├── a13_a10s_r8.c
│   ├── a20.c
│   ├── a23.c
│   ├── a31.c
│   ├── a33_r16.c
│   ├── a40i_r40.c
│   ├── a64.c
│   ├── a80.c
│   ├── a83t.c
│   ├── d1_f133.c
│   ├── f1c100s_f1c200s_f1c500s.c
│   ├── h2_h3.c
│   ├── h5.c
│   ├── h6.c
│   ├── h616.c
│   ├── r328.c
│   ├── r329.c
│   ├── t507.c
│   ├── v3s_s3.c
│   └── v831.c
├── fel.c
├── fel.h
├── main.c
└── payloads
    ├── d1_f133
    │   ├── ddr
    │   │   ├── Makefile
    │   │   ├── include
    │   │   │   ├── byteorder.h
    │   │   │   ├── d1
    │   │   │   │   └── reg-ccu.h
    │   │   │   ├── endian.h
    │   │   │   ├── io.h
    │   │   │   ├── linkage.h
    │   │   │   ├── riscv64.h
    │   │   │   ├── stdarg.h
    │   │   │   ├── stddef.h
    │   │   │   ├── stdint.h
    │   │   │   ├── string.h
    │   │   │   ├── types.h
    │   │   │   └── xboot.h
    │   │   ├── link.ld
    │   │   └── source
    │   │       ├── mctl_hal.S
    │   │       ├── memcpy.S
    │   │       ├── memset.S
    │   │       ├── start.S
    │   │       ├── sys-clock.c
    │   │       ├── sys-dram.c
    │   │       └── sys-uart.c
    │   ├── jtag
    │   │   ├── Makefile
    │   │   ├── include
    │   │   │   ├── byteorder.h
    │   │   │   ├── d1
    │   │   │   │   └── reg-ccu.h
    │   │   │   ├── endian.h
    │   │   │   ├── io.h
    │   │   │   ├── linkage.h
[... 565 lines cut ...]
        │   │   ├── types.h
        │   │   ├── v831
        │   │   │   ├── reg-ccu.h
        │   │   │   └── reg-dram.h
        │   │   └── xboot.h
        │   ├── link.ld
        │   └── source
        │       ├── start.S
        │       └── sys-jtag.c
        └── spi
            ├── Makefile
            └── include
                ├── byteorder.h
                ├── endian.h
                ├── io.h
                ├── stdarg.h
                ├── stddef.h
                ├── stdint.h
                ├── string.h
                ├── types.h
                └── v831

70 directories, 290 files

```

jianjunjiang, 2021-11-09:
缺少什么文件，编译时提示的吗？

taotieren, 2021-11-09:
[image]
是的，编译的时候提示，发行包内的文件数量和 master 分支的文件数量不一致。

taotieren, 2021-11-09:
[image]
这些文件在发行版里面是没有的，是没有加入进去吗？

jianjunjiang, 2021-11-09:
没有这个异常，估计你的脚本哪里出问题了，你可以直接reset某个tag查看，文件都存在的。

closed 2021-11-11 by jianjunjiang

taotieren, 2021-11-11:
可能是使用代理站下载的 `https://ghproxy.com/https://github.com/xboot/xfel/archive/refs/tags/v1.2.1.tar.gz` 的文件有问题，刚才重试发现正常。