xboot/xfel #27: Error: usb bulk send error (operation timed out)

duskmoon314, 2022-03-20:
I tried to write a binary to a D1S board and an R528 board and encountered a `usb bulk send error`. To know what happened, I modified the source code to print the error message.

I execute commands like below:

```console
> xfel version
AWUSBFEX ID=0x00185900(D1/F133) dflag=0x44 dlength=0x08 scratchpad=0x00045000
> xfel ddr d1
Initial ddr controller succeeded
> xfel write 0x80000000 target/riscv64gc-unknown-none-elf/debug/blink.bin
usb bulk send error: Operation timed out

> xfel version
AWUSBFEX ID=0x00185900(R528/T113) dflag=0x44 dlength=0x08 scratchpad=0x00045000
> xfel ddr r528-s3
Initial ddr controller succeeded
> xfel write 0x80000000 target/armv7a-none-eabi/debug/blink.bin
usb bulk send error: Operation timed out
```

Serval weeks ago, I used `xfel ddr ddr2` to initialize the D1S board and it worked fine. But now I can't get it to work.

What could be causing this error? How do I fix it?

---

我尝试向D1S和R528的板子写入二进制文件，但遇到了`usb bulk send error`。我修改了源代码来尝试找出问题，下面是我执行的命令：

```console
# 如上
```

几周前我还可以用`xfel ddr ddr2`来初始化D1S的那块板子并运行二进制文件。但现在不行了。

可能是什么导致的？如何解决呢？

duskmoon314, 2022-03-21:
又做了一些尝试，发现在 `xfel ddr r528-s3` 之后，板子似乎直接退出了fel模式，导致之后的指令都无法运行

```console
> xfel ddr r528-s3
Initial ddr controller succeeded
> xfel version
usb bulk send error: Operation timed out
```

相较之下，正常的 d1s 是这样的：

```console
> xfel version
AWUSBFEX ID=0x00185900(D1/F133) dflag=0x44 dlength=0x08 scratchpad=0x00045000
> xfel ddr f133
Initial ddr controller succeeded
> xfel version
AWUSBFEX ID=0x00185900(D1/F133) dflag=0x44 dlength=0x08 scratchpad=0x00045000
```

jianjunjiang, 2022-03-22:
看操作命令是正确的，你这芯片确认是r528-s3码？有没有可能是r528-s4或者r528-s2，这些ddr不一致的，当然初始化参数也会不一样，当然默认参数是528-s3的，你这个无法识别，感觉payload没能成功运行。

duskmoon314, 2022-03-22:
> 你这芯片确认是r528-s3码？有没有可能是r528-s4或者r528-s2，这些ddr不一致的

我目前只知道是 r528，有什么办法可以确认是几代吗？ddr的payload要如何得到对应的呢？我看代码中都是二进制硬编码的。

jianjunjiang, 2022-03-22:
https://github.com/xboot/xfel/blob/master/chips/r528_t113.c#L4158

主要是填充struct ddr3_param_t ddr3这个结构体，如果你能获取你芯片的ddr正确配置参数，可以填进来试试。

我现在还没有能找到如何区分各种马甲的方法，全志应该有一套编码规则，用来标识各种封装的马甲。逆向了一部分闭源代码，如果对比不同马甲的这部分代码，应该能找到区别。也很有可能分析出全志的编码规则，不过还是需要多收集足够多的细节才行。

```
uint id_judge_fun(uint param_1)

{
  uint uVar1;
  
  if ((_DAT_01c15004 & 7) == 0) {
    uVar1 = disturb_coding(param_1,1);
    return uVar1;
  }
  return 0;
}

uint disturb_coding(uint param_1,int param_2)

{
  int iVar1;
  
  iVar1 = crc_16(param_1,param_2);
  return param_1 + param_2 | iVar1 << 0x10;
}

void crc_16(uint param_1,int param_2)

{
  uint uVar1;
  uint uVar2;
  short sVar3;
  
  uVar1 = param_1 ^ param_2 << 8;
  sVar3 = 8;
  do {
    uVar2 = uVar1 & 0x8000;
    uVar1 = (uVar1 & 0x7fff) << 1;
    if (uVar2 != 0) {
      uVar1 = (uVar1 ^ 0xffff8005) & 0xffff;
    }
    sVar3 = sVar3 + -1;
  } while (sVar3 != 0);
  return;
}
```

duskmoon314, 2022-03-23:
我找到了一段参数，用于填充 `ddr3_param_t`。我加了一条指令 `xfel ddr test` 用这段参数进行填充，不过似乎无论是 r528 还是 t113 的 payload 加载之后，虽然仍可以执行 `xfel version` 之类的指令，但是写入我写的二进制文件后执行不成功。我不确定能不能完全判断是我的二进制文件有误，还是 payload 也需要找一个进行填充？

jianjunjiang, 2022-03-30:
能正常运行，会有串口信息的，首先还是确认你手上的芯片是那一颗，看是否是新马甲，如果是新马甲，就需要支持一下了。

jianjunjiang, 2022-03-30:
> 

你可以读写DDR空间来确认是否DDR是否已正常初始化

closed 2022-04-15 by jianjunjiang