xboot/xfel #29: xfel擦除对齐地址时会多擦除一个扇区

0helong0, 2022-05-18:
使用xfel擦除（nor）和烧写（nor）地址容量都是对齐时，会多擦除一个扇区，这样会擦除掉后面的分区数据：
```sh
xfel spinor erase 0x00 0x00200000
100% [================================================] 2.004 MB, 343.741 KB/s
xfel spinor erase 0x00 0x00100000
100% [================================================] 1.004 MB, 336.865 KB/s
xfel spinor erase 0x00 0x0080000
100% [================================================] 516.000 KB, 327.770 KB/s
```
目前擦除（856行）和烧写（916行）的容量计算为：
```c
cnt = ((addr & emask) + len + esize) & ~emask;  // 计算对齐的地址和容量时，会多计算一个<esize>
```
是否可以改为：
```c
cnt = ((addr & emask) + len - 1 + esize) & ~emask; 
```

jianjunjiang, 2022-05-18:
减1的话，如果addr本省就死后对齐的，len正好是一个erase size，那计算的cnt是不是就少了一呢。这个需要计算分析下，看看各个边界情况。

jianjunjiang, 2022-06-02:
https://github.com/xboot/xfel/pull/31
参见这个提交，看是否解决了问题。不知还有没有bug，需要多多测试。

aozima, 2022-06-02:
原来这还有个 issue，早没看到。这个PR后代码还得再改改。我先试着改下。

PR #32 (2022-06-02): 修正擦除起止范围计算
修正擦除起止范围计算，计算方法简化为： align_end - align_start

fixed #29 

关联 #31
closed 2022-06-02 by jianjunjiang