xboot/xfel #64: T113-S4 ddr initialization

robots, 2024-06-30:
Hi,

T113-s4 ddr will not initialize with existing T113-s3 code.

Console will read:
```
[1753]fes begin commit:4d16602
[1755]set pll start
[1757]fix vccio detect value:0xc0
[1764]periph0 has been enabled
[1768]set pll end
[1769][pmu]: bus read error
[1772]board init ok
[1774]beign to init dram
[1776]ZQ value = 0x2f
[1778]get_pmu_exist() = -1
[1780]ddr_efuse_type: 0xa
[1783]trefi:7.8ms
[1785]DX0 state:3
[1787]DX1 state:3
[1789][ERROR DEBUG] auto scan dram rank&width fail !
```

According to this page, we need to change the ac remapping:
https://bbs.aw-ol.com/topic/4102/%E8%AF%B7%E9%97%AE%E8%B0%81%E6%9C%89aic600e3%E7%9A%84dram%E5%88%9D%E5%A7%8B%E5%8C%96%E4%BB%A3%E7%A0%81/4?lang=en-US

```
diff --git a/chips/r528_t113.c b/chips/r528_t113.c
index 79159c0..6f85489 100644
--- a/chips/r528_t113.c
+++ b/chips/r528_t113.c
@@ -4283,8 +4283,20 @@ static int chip_ddr(struct xfel_ctx_t * ctx, const char * type)
                                .dram_tpr12 = 0x00000046,
                                .dram_tpr13 = 0x34000100,
                        };
+
+                       // patch if t113-s4 and fuse = 0xa: select ac_remap[0]
+                       //                              switchD_0002bd18::caseD_a                       XREF[1]:     0002bd18(j)
+                       //  0002bd4a 16 22           movs       r2,#0x16
+                       //  0002bd4c 1e a9           add        r1,sp,#0x78   <- points to ac_remap[5] on stack, change to ac_remap[0] on stack (offset 0)
+                       //  0002bd4e eb e7           b          LAB_0002bd28
+                       // after patch
+                       //  0002bd4c 00 a9           add        r1,sp,#0x00
+
+                       uint8_t patch[] = { 0x00, 0xa9 };
+
                        fel_write(ctx, 0x00028000, (void *)&t113_ddr_payload[0], sizeof(t113_ddr_payload));
                        fel_write(ctx, 0x00028038, (void *)&ddr3, sizeof(ddr3));
+                       fel_write(ctx, 0x0002bd4c, (void *)&patch[0], sizeof(patch));
                        fel_exec(ctx, 0x00028000);
                        return 1;
                }
```

With this patch in place, i can successfully initialize ddr memory.

```
[2932]fes begin commit:4d16602
[2935]set pll start
[2937]fix vccio detect value:0xc0
[2944]periph0 has been enabled
[2947]set pll end
[2949][pmu]: bus read error
[2952]board init ok
[2953]beign to init dram
[2956]ZQ value = 0x2f
[2958]get_pmu_exist() = -1
[2960]ddr_efuse_type: 0xa
[2963]trefi:7.8ms
[2965][AUTO DEBUG] single rank and full DQ!
[2969]ddr_efuse_type: 0xa
[2971]trefi:7.8ms
[2974][AUTO DEBUG] rank 0 row = 14
[2977][AUTO DEBUG] rank 0 bank = 8
[2980][AUTO DEBUG] rank 0 page size = 2 KB
[2984]DRAM BOOT DRIVE INFO: V0.33
[2987]DRAM CLK = 936 MHz
[2990]DRAM Type = 3 (2:DDR2,3:DDR3)
[2993]DRAMC read ODT  off.
[2995]DRAM ODT value: 0x42.
[2998]ddr_efuse_type: 0xa
[3001]DRAM SIZE =256 M
[3003]dram_tpr4:0x0
[3005]PLL_DDR_CTRL_REG:0xf8004d00
[3008]DRAM_CLK_REG:0xc0000000
[3011][TIMING DEBUG] MR2= 0x20
[3015]DRAM simple test OK.
[3017]rtc standby flag is 0x0, super standby flag is 0x0
[3022]init dram ok
```

ua1arn, 2024-08-02:
I can confirm - remap in style 0 succesfully work.
Init params also updated.

jianjunjiang, 2025-05-01:
已经支持

closed 2025-05-01 by jianjunjiang