xboot/xfel #70: V851S JTAG on PH11-14 ARM CPU

voloviq, 2025-01-29:
Hi,

I try to force my board with v851s to start debugging on port PH11 up to PH14. I made a few experiments

1. Go to the directory payload and modify sys_jtag_init(void) to 

`void sys_jtag_init(void)
{
	virtual_addr_t addr;
	u32_t val;

	/* Config GPIOH11, GPIOH12, GPIOH13 and GPIOH14 to JTAG mode */
	addr = 0x02000154 + 0x00;

	val = read32(addr);
	val &= ~(0xf << ((11 & 0x7) << 2));
	val |= ((0x2 & 0xf) << ((11 & 0x7) << 2));
	write32(addr, val);

	val = read32(addr);
	val &= ~(0xf << ((12 & 0x7) << 2));
	val |= ((0x2 & 0xf) << ((12 & 0x7) << 2));
	write32(addr, val);

	val = read32(addr);
	val &= ~(0xf << ((13 & 0x7) << 2));
	val |= ((0x2 & 0xf) << ((13 & 0x7) << 2));
	write32(addr, val);

	val = read32(addr);
	val &= ~(0xf << ((14 & 0x7) << 2));
	val |= ((0x2 & 0xf) << ((14 & 0x7) << 2));
	write32(addr, val);
}`

then write to memory using

`xfel write 0x30000 jtag.bin`
`xfel exec 0x30000`

no handling on JTAG_DO.

I'm not sure if GPIOH activation is necessary.

2. I try to use the suggested code

`xfel jtag`

also zero reaction on JTAG_DO

Unfortunately in the file in function "static int chip_jtag(...." magic numbers are only present. So I don't know how to change the port from PF (I guess it's default) to PH. 

Can You please give me some hints.

Thanks in advanced
BR
Michal

YuzukiTsuru, 2025-01-30:
Set PH mux reg 0x02000154 to 0x22222222 and PF set to 0x020000F0 set to 0x0FFFFFFF then try again.

voloviq, 2025-02-03:
Hello,

sorry for reopen this issue but I can't debug my elf file. The openocd seems to be connected well but I can't debug, loaded program didn't start.

I prepared a simple blinking LED program. When I write it to v851s using xfel

```
xfel write 0x30000 awboot-fel.bin
xfel exec 0x30000
```
the LED bilnking as expected.

but when I start this program using

```
gdb-multiarch awboot-fel.elf
target extended-remote localhost:3333
monitor reset halt
load
n
```

the program didn't start and I get an error.

I attach gdb log and openocd log

```
Reading symbols from awboot-fel.elf...
(gdb) target extended-remote localhost:3333
Remote debugging using localhost:3333
0x00008860 in ?? ()
(gdb) monitor reset halt
JTAG tap: v851s.dap tap/device found: 0x5ba00477 (mfg: 0x23b (ARM Ltd), part: 0xba00, ver: 0x5)
v851s.cpu rev 5, partnum c07, arch f, variant 0, implementor 41
target halted in ARM state due to debug-request, current mode: Supervisor
cpsr: 0x60000153 pc: 0x00008860
MMU: disabled, D-Cache: disabled, I-Cache: enabled
(gdb) load
Loading section .text, size 0x186c lma 0x30000
Loading section .rodata.show_regs.str1.1, size 0x67 lma 0x3186c
Loading section .rodata.arm32_do_undefined_instruction.str1.1, size 0x9 lma 0x318d3
Loading section .rodata.__FUNCTION__.0, size 0x14 lma 0x318dc
Loading section .rodata.__FUNCTION__.1, size 0x18 lma 0x318f0
Loading section .rodata.__FUNCTION__.2, size 0x1c lma 0x31908
Loading section .rodata.__FUNCTION__.3, size 0x1f lma 0x31924
Loading section .rodata.sunxi_clk_init.str1.1, size 0x5b lma 0x31943
Loading section .rodata.formatStates, size 0x5b lma 0x3199e
Loading section .rodata.ms_digits, size 0x11 lma 0x319f9
Loading section .rodata.ms_false, size 0x6 lma 0x31a0a
Loading section .rodata.ms_true, size 0x5 lma 0x31a10
Loading section .rodata.ms_null, size 0x7 lma 0x31a15
Loading section .data.led_blue, size 0x4 lma 0x31a1c
Loading section .data.usart_dbg, size 0x12c lma 0x31a20
Loading section .ARM.exidx, size 0x8 lma 0x31b4c
--Type <RET> for more, q to quit, c to continue without paging--
Start address 0x00030000, load size 6996
Transfer rate: 8 KB/sec, 437 bytes/write.
(gdb) n
Single stepping until exit from function reset,
which has no line number information.
v851s.cpu rev 5, partnum c07, arch f, variant 0, implementor 41
v851s.cpu rev 5, partnum c07, arch f, variant 0, implementor 41
```
and openocd log

```
Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : J-Link V11 compiled Sep  3 2024 10:40:48
Info : Hardware version: 11.00
Info : VTarget = 3.309 V
Info : clock speed 1000 kHz
Info : JTAG tap: v851s.dap tap/device found: 0x5ba00477 (mfg: 0x23b (ARM Ltd), part: 0xba00, ver: 0x5)
Info : v851s.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for v851s.cpu on 3333
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
Info : v851s.cpu rev 5, partnum c07, arch f, variant 0, implementor 41
Info : v851s.cpu: MPIDR level2 0, cluster 0, core 0, multi core, no SMT
target halted in ARM state due to debug-request, current mode: Supervisor
cpsr: 0x60000153 pc: 0x00008860
MMU: disabled, D-Cache: disabled, I-Cache: enabled
Info : JTAG tap: v851s.dap tap/device found: 0x5ba00477 (mfg: 0x23b (ARM Ltd), part: 0xba00, ver: 0x5)
Info : v851s.cpu rev 5, partnum c07, arch f, variant 0, implementor 41
target halted in ARM state due to debug-request, current mode: Supervisor
cpsr: 0x60000153 pc: 0x00008860
MMU: disabled, D-Cache: disabled, I-Cache: enabled
Info : v851s.cpu rev 5, partnum c07, arch f, variant 0, implementor 41
Info : v851s.cpu rev 5, partnum c07, arch f, variant 0, implementor 41
Error: timeout waiting for DSCR bit change
Error: Error waiting for InstrCompl=1
Error: Error waiting for cortex_a_exec_opcode
Error: timeout waiting for DSCR bit change
Error: Error waiting for InstrCompl=1
Error: timeout waiting for DSCR bit change
Error: Error waiting for InstrCompl=1
Error: Error waiting for cortex_a_exec_opcode
Warn : keep_alive() was not invoked in the 1000 ms timelimit. GDB alive packet not sent! (3017 ms). Workaround: increase "set remotetimeout" in GDB
```

In the internet I found some information that I should add something to my code.

Thanks in advanced
BR Michał

voloviq, 2025-02-04:
I think I found a solution it is necessary to configure PH JTAG pins into debugged the application.

BR
Michal

closed 2025-02-04 by voloviq